osrf/ros2_test_cases #966: FastDDS Configuration

Yadunund, 2023-05-01:
Check the documentation for the 'FastDDS Configuration' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [FastDDS Configuration page](https://docs.ros.org/en/iron/Tutorials/FastDDS-Configuration/FastDDS-Configuration.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

iuhilnehc-ynos, 2023-05-15:
* [x]  **I was able to follow the documentation.**
* [x]  **The documentation seemed clear to me.**
* [x]  **The documentation didn't have any obvious errors.**

Found a log missing a period in the doc.
```
[INFO] [1612977403.805799037] [ping_server]: Ready to serve
```

<details><summary> log </summary>
<p>

- Limiting the number of matching subscribers

[image]

- Using partitions within the topic

[image]

- Configuring a service and a client

[image]

</p>
</details>

Yadunund, 2023-05-15:
Awesome, thanks for testing!

closed 2023-05-15 by Yadunund