osrf/ros2_test_cases #967: Gazebo

Yadunund, 2023-05-01:
Check the documentation for the 'Gazebo' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Gazebo page](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Gazebo.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

klu17, 2023-05-09:
Was able tutorial up to '2 Configuring ROS 2'. At the point, I got an error: `Package 'ros_gz_bridge' not found` and was not able to find `ros_gz_bridge` install instructions for iron at [ROS Index]( https://index.ros.org/p/ros_gz_bridge/).

[image]
[image]
[image]

Yadunund, 2023-05-11:
Thanks for trying this. Turns out we're still in the process of releasing debians for [ros_gz](https://github.com/gazebosim/ros_gz) which is why you're unable to install it at the moment. Will update once that's ready.

clalancette, 2023-06-23:
These packages weren't available, but should be available now.  With the tutorial party over, I'm going to close this out.  Thanks for your efforts here.

closed 2023-06-23 by clalancette