osrf/ros2_test_cases #968: Getting Started With Ros2doctor

Yadunund, 2023-05-01:
Check the documentation for the 'Getting Started With Ros2doctor' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Getting Started With Ros2doctor page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Getting-Started-With-Ros2doctor.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-05:
Taking this

marioprats, 2023-05-05:
All good!

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette