osrf/ros2_test_cases #969: Ignition

Yadunund, 2023-05-01:
Check the documentation for the 'Ignition' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Ignition page](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Ignition.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
Unable to complete steps after [2 Configuring ROS2](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Gazebo.html#configuring-ros-2). Similar to another issue I tested, installing ros_gz_bridge is a blocking step.

[image]

Yadunund, 2023-05-18:
This ticket should not have been created as it's a duplicate of https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Gazebo.html. It probably got created due to a redirect link in the website. Will close this one and follow up in the other.

closed 2023-05-18 by Yadunund