osrf/ros2_test_cases #970: Intermediate

Yadunund, 2023-05-01:
Check the documentation for the 'Intermediate' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Intermediate page](https://docs.ros.org/en/iron/Tutorials/Intermediate.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

christophebedard, 2023-05-16:
@Bi0T1N you mentioned this issue in https://github.com/ros2/ros2_documentation/pull/3526, but I'm not sure if that is really related to this issue. Is it related?

christophebedard, 2023-05-16:
Otherwise, there isn't much to validate here; the list of tutorials looks good and matches the sidebar.

clalancette, 2023-05-16:
Thanks for testing!  I'm going to close this out because I agree that it isn't much related to the `ros2_documentation` PR.

closed 2023-05-16 by clalancette

Bi0T1N, 2023-05-17:
> @Bi0T1N you mentioned this issue in [ros2/ros2_documentation#3526](https://github.com/ros2/ros2_documentation/pull/3526), but I'm not sure if that is really related to this issue. Is it related?

Its partly related but the path was fixed already. Other changes are additional...