osrf/ros2_test_cases #971: Intra Process Communication

Yadunund, 2023-05-01:
Check the documentation for the 'Intra Process Communication' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Intra Process Communication page](https://docs.ros.org/en/iron/Tutorials/Demos/Intra-Process-Communication.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-10:
Assigned to myself.

Can successfully confirm the follwoing:
[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

Yadunund, 2023-05-10:
Thanks for testing!

closed 2023-05-10 by Yadunund