osrf/ros2_test_cases #972: Introducing ros2 security

Yadunund, 2023-05-01:
Check the documentation for the 'Introducing ros2 security' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Introducing ros2 security page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Introducing-ros2-security.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

klu17, 2023-05-09:
I'll take this one. I was able to follow through the tutorial and receive messages on the listener:

[image]

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette