osrf/ros2_test_cases #973: Introducing Turtlesim

Yadunund, 2023-05-01:
Check the documentation for the 'Introducing Turtlesim' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Introducing Turtlesim page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

sea-bass, 2023-05-03:
Worked great for me!

clalancette, 2023-05-03:
Thanks for testing!

closed 2023-05-03 by clalancette