osrf/ros2_test_cases #974: Introduction To Tf2

Yadunund, 2023-05-01:
Check the documentation for the 'Introduction To Tf2' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Introduction To Tf2 page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-06:
On a jammy debian based installation.

I was able to follow the documentation, the documentation was clear to me and did not contained any obvious error.

Only slight question/improvment is in the _**Using tf2_echo**_ section, the output seems now to be more detailed than in the documentation, so could be updated ? 

Here is an example of my output with `iron`:

```
At time 1683377979.902062451
- Translation: [0.963, 0.172, 0.000]
- Rotation: in Quaternion [0.000, 0.000, -0.079, 0.997]
- Rotation: in RPY (radian) [0.000, 0.000, -0.158]
- Rotation: in RPY (degree) [0.000, 0.000, -9.052]
- Matrix:
  0.988  0.157  0.000  0.963
 -0.157  0.988 -0.000  0.172
 -0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
At time 1683377980.909846037
- Translation: [0.485, 0.038, 0.000]
- Rotation: in Quaternion [0.000, 0.000, -0.169, 0.986]
- Rotation: in RPY (radian) [0.000, 0.000, -0.339]
- Rotation: in RPY (degree) [0.000, 0.000, -19.415]
- Matrix:
  0.943  0.332  0.000  0.485
 -0.332  0.943 -0.000  0.038
 -0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
```

```

clalancette, 2023-05-06:
> Only slight question/improvment is in the _**Using tf2_echo**_ section, the output seems now to be more detailed than in the documentation, so could be updated ?

Yes, that's a good point.  See https://github.com/ros2/ros2_documentation/pull/3538 .

Since the testing was otherwise successful, I'm going to close this out.  Thanks for testing!

closed 2023-05-06 by clalancette