osrf/ros2_test_cases #975: Introspection with command line tools

Yadunund, 2023-05-01:
Check the documentation for the 'Introspection with command line tools' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Introspection with command line tools page](https://docs.ros.org/en/iron/Tutorials/Introspection-with-command-line-tools.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

nirwester, 2023-05-09:
Working and clear documentation, couldn't self-assign.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette