osrf/ros2_test_cases #976: Launch files migration guide

Yadunund, 2023-05-01:
Check the documentation for the 'Launch files migration guide' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Launch files migration guide page](https://docs.ros.org/en/iron/Tutorials/Launch-files-migration-guide.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-10:
Assigned to myself.

Can successfully confirm the following:
[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

Yadunund, 2023-05-10:
Thanks for testing!

closed 2023-05-10 by Yadunund