osrf/ros2_test_cases #977: Learning About Tf2 And Time Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Learning About Tf2 And Time Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Learning About Tf2 And Time Cpp page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

I have been able to follow the documentation, it was clear to me, but there is slight inaccuracies in the code description in regards to the current state of the `turtle_tf2_listener.cpp` file created in previous tutorials.

1. I would update the 3 pieces of code that reference `lookupTransform` to match the updated version of `turtle_tf2_listener.cpp` that now uses `t=` and a `try catch`
Namely replace with :
```
        try {
          t = tf_buffer_->lookupTransform(
            toFrameRel, fromFrameRel,
            tf2::TimePointZero);
        } catch (const tf2::TransformException & ex) {
            ...
```
```
        rclcpp::Time now = this->get_clock()->now();
        try {
          t = tf_buffer_->lookupTransform(
            toFrameRel, fromFrameRel,
            now);
        } catch (const tf2::TransformException & ex) {
           ...
```
and
```
        rclcpp::Time now = this->get_clock()->now();
        try {
          t = tf_buffer_->lookupTransform(
            toFrameRel, fromFrameRel,
            now,
            50ms);
        } catch (const tf2::TransformException & ex) {
         ...
```
2. The tutorial misses the information to 'recompile'  before each suggestion to launch on modified code.
So I would just replace in section __"1. tf2 and time"__
`Now try to run the launch file.`
with
`Now compile the package and try to run the launch file.`

and replace in section __"3. Checking the results"__
`You can now run the launch file.`
By
`You can now compile the package and run the launch file.`

Yadunund, 2023-05-10:
Thanks for testing and the feedback! Could you open a PR on ros2/ros2_documentation with these changes?

Yadunund, 2023-05-11:
I'll close there is a PR to update the doc.

closed 2023-05-11 by Yadunund