osrf/ros2_test_cases #978: Learning About Tf2 And Time Py

Yadunund, 2023-05-01:
Check the documentation for the 'Learning About Tf2 And Time Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Learning About Tf2 And Time Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy with fastDDS.

**_ERROR_**: This tutorial page is outdated.

To be precise, the argument it wants to show off still exists `timeout=` but the code of the `turtle_tf2_listener.py` it proposes to modify has now been changed.
So the _"Update the listener node"_ section is inapplicable, the `lookup_transform` call is now in the `on_timer` method which is called every seconds, and first spawn and wait for spawning of the turtle before starting to look for the transform, so it never fails when reaching the `lookup_transform` call.

I have tried to find several simple ways to modify the code to  make this tutorial page work again, but I have not find anything compelling and not fully artificial.

Note that this tutorial page is outdated since Foxy at least. (oldest version I could find on docs.ros.org)

BorisBoutillier, 2023-05-09:
Now that I have done the C++ version of this tutorial, and the back in time PY page, I understand a bit more the difference between ros.time.Time() and self.get_clock().now().

And I would have expected that the modified working code for this tutorial would have been:
```
                now = self.get_clock().now()
                try:
                    t = self.tf_buffer.lookup_transform(
                        to_frame_rel,
                        from_frame_rel,
                        now,
                        timeout=rclpy.duration.Duration(seconds=0.05))
                except TransformException as ex:
                        ....
```
But adding a `timeout=` does not change anything to the behaviour of the code, error about "extrapolation in future" still pops up, while in the C++ version, adding the timeout behaves as described.
The extrapolation errors still pops with a 1s timeout.

No clue if this would have been the correct modification, and why it does not work with the python version. Can clearly be an error on my side.

Yadunund, 2023-05-11:
Thanks for trying this out. Could we update your existing PR with these changes? I'll close this ticket.

closed 2023-05-11 by Yadunund

BorisBoutillier, 2023-05-11:
Unfortunately this time I don't know how to fix it, the way I though it should work in pyhon does not work, and is not helpful to the tutorial. So I prefer to let it as is, instead of modify it to a " not more working" version :)