osrf/ros2_test_cases #979: Logging and logger configuration

Yadunund, 2023-05-01:
Check the documentation for the 'Logging and logger configuration' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Logging and logger configuration page](https://docs.ros.org/en/iron/Tutorials/Demos/Logging-and-logger-configuration.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jacobperron, 2023-05-08:
I'm testing this 👷

jacobperron, 2023-05-08:
Tutorial and code LGTM!

clalancette, 2023-05-08:
Rock on Jacob!

Thanks for testing.

closed 2023-05-08 by clalancette