osrf/ros2_test_cases #980: Managed Nodes

Yadunund, 2023-05-01:
Check the documentation for the 'Managed Nodes' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Managed Nodes page](https://docs.ros.org/en/iron/Tutorials/Demos/Managed-Nodes.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
I'll take this one.

karmenlu, 2023-05-09:
- Able to run demo with three live terminals
- Able to launch demo with one launch file in one terminal
- Got current state of `lc_talker` node:
[image]
- Output for executing a state change differs from expected snippet on instruction page
[image]
- List of currently available states differs from snippet on instruction page
[image]
- Able to run steps with `ros2 service`