osrf/ros2_test_cases #981: Miscellaneous

Yadunund, 2023-05-01:
Check the documentation for the 'Miscellaneous' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Miscellaneous page](https://docs.ros.org/en/iron/Tutorials/Miscellaneous.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

christophebedard, 2023-05-16:
There isn't much to validate here; the list of tutorials looks good and matches the sidebar.

clalancette, 2023-05-16:
Thanks for testing!

closed 2023-05-16 by clalancette