osrf/ros2_test_cases #984: Overriding QoS Policies For Recording And Playback

Yadunund, 2023-05-01:
Check the documentation for the 'Overriding QoS Policies For Recording And Playback' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Overriding QoS Policies For Recording And Playback page](https://docs.ros.org/en/iron/Tutorials/Ros2bag/Overriding-QoS-Policies-For-Recording-And-Playback.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

OmkarKabadagi5823, 2023-05-12:
I will take this up

OmkarKabadagi5823, 2023-05-13:
The documentation was clear and all the commands given ran successfully as intended.

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

## Result
### Publisher
```bash
ros2 topic pub -r 0.1 --qos-durability transient_local /talker std_msgs/String "data: Hello World"
```
```
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='Hello World')

publishing #2: std_msgs.msg.String(data='Hello World')

publishing #3: std_msgs.msg.String(data='Hello World')

publishing #4: std_msgs.msg.String(data='Hello World')

publishing #5: std_msgs.msg.String(data='Hello World')
```

### Rosbag record (Recording started after 3 messages were already published)
```bash
ros2 bag record -a -o my_bag --qos-profile-overrides-path durability_override.yaml
```
```
[INFO] [1683995286.860967866] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1683995286.863010103] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1683995286.863585363] [rosbag2_recorder]: Listening for topics...
[INFO] [1683995286.864335514] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1683995286.864859447] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1683995286.966566952] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1683995286.967008523] [rosbag2_recorder]: Overriding subscription profile for /talker
[INFO] [1683995286.968534348] [rosbag2_recorder]: Subscribed to topic '/talker'
[INFO] [1683995313.102381877] [rclcpp]: signal_handler(signum=2)
[INFO] [1683995313.159655334] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1683995313.167951842] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
```

### Rosbag play
```bash
ros2 bag play --qos-profile-overrides-path reliability_override.yaml my_bag
```
```
[INFO] [1683995332.700701948] [rosbag2_player]: Set rate to 1
[INFO] [1683995332.700805111] [rosbag2_player]: Overriding QoS profile for topic /talker
[WARN] [1683995332.701749691] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT
[INFO] [1683995332.706135926] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1683995332.706166583] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1683995332.706181982] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1683995332.706196610] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1683995332.706210507] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1683995332.706281054] [rosbag2_player]: Playback until timestamp: -1
```

### Rostopic echo (Received all messages as expected with `history: keep_all`)
```bash
 ros2 topic echo --qos-reliability best_effort /talker std_msgs/String
```
```
data: Hello World
---
data: Hello World
---
data: Hello World
---
data: Hello World
---
data: Hello World
---
```

clalancette, 2023-05-13:
Thank you for testing!

closed 2023-05-13 by clalancette