osrf/ros2_test_cases #985: Parameters YAML files migration guide

Yadunund, 2023-05-01:
Check the documentation for the 'Parameters YAML files migration guide' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Parameters YAML files migration guide page](https://docs.ros.org/en/iron/Tutorials/Parameters-YAML-files-migration-guide.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-09:
Assigned to myself.

I can successfully confirm the following checks:

[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette