osrf/ros2_test_cases #986: Pluginlib

Yadunund, 2023-05-01:
Check the documentation for the 'Pluginlib' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Pluginlib page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Pluginlib.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-05:
Taking this

fujitatomoya, 2023-05-06:
@marioprats good eye, that generates the error if user just adds the lines as described.

```bash
root@raspi4-1:~/docker_ws/iron_ws# colcon build --packages-select polygon_base polygon_plugins
Starting >>> polygon_base
Finished <<< polygon_base [19.1s]
Starting >>> polygon_plugins
--- stderr: polygon_plugins
CMake Error at CMakeLists.txt:34 (add_library):
  add_library cannot create target "polygon_plugins" because another target
  with the same name already exists.  The existing target is a shared library
  created in source directory "/root/docker_ws/iron_ws/src/polygon_plugins".
  See documentation for policy CMP0002 for more details.

---
Failed   <<< polygon_plugins [3.93s, exited with code 1]

Summary: 1 package finished [23.9s]
  1 package failed: polygon_plugins
  1 package had stderr output: polygon_plugins
```

> I'll send a PR if that's okay. Should I update just iron, or rolling?

rolling, and then we can backport to iron.

fujitatomoya, 2023-05-06:
/assign @fujitatomoya , i will confirm this including aarch64 platform.

fujitatomoya, 2023-05-06:
except https://github.com/osrf/ros2_test_cases/issues/986#issuecomment-1536590834 @marioprats pointed out, all procedure has no problem for both amd64 and aarch64.

clalancette, 2023-05-07:
> I'll send a PR if that's okay. Should I update just iron, or rolling?

Please do, and target `rolling`.  From there we'll backport to `iron`.  Once that PR is open, we can close this issue out.

marioprats, 2023-05-08:
PR open here: https://github.com/ros2/ros2_documentation/pull/3544

clalancette, 2023-05-08:
Thank you for the PR!  I've now merged it, so with that done I'm going to call this one done.

Thank you for testing!

closed 2023-05-08 by clalancette