osrf/ros2_test_cases #987: Py

Yadunund, 2023-05-01:
Check the documentation for the 'Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-08:
On a debian jammy installation with fastDDS.

I was followed along the action tutorial pages, and this ' write an action server-client' py page.

I was able to follow the documentation, including writing and executing the proposed code. The documentation was clear to me, I was able to have everything running. I have not seen any obvious error, neither in wording, code or linked pages.

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette