osrf/ros2_test_cases #991: Real Time Programming

Yadunund, 2023-05-01:
Check the documentation for the 'Real Time Programming' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Real Time Programming page](https://docs.ros.org/en/iron/Tutorials/Demos/Real-Time-Programming.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-11:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

1. At rttest_plot section, The ```rttest_plot pendulum_demo_results``` is not available as standalone command. But it runs with ```ros2 run rttest rttest_plot pendulum_demo_results```.
2. The ```ulimit -l unlimited``` command require root privileges (with sudo it didn't work). First need to login to root user with ```sudo su``` command and execute ```ulimit -l unlimited``` command. Then exit from root user with ```exit``` command and continue with the documentation.
3. The rttest_plot script generates 4 SVG file not 3 as said in documentation. They are 

- pendulum_demo_results_plot_latency.svg
- pendulum_demo_results_plot_latency_hist.svg
- pendulum_demo_results_plot_majflts.svg
- pendulum_demo_results_plot_minflts.svg

The pendulum_demo_results_plot_latency_hist.svg is missing from the documentation.

**Other-than these issue I am able to follow the document and execute the demo.**

clalancette, 2023-05-15:
@pradyum I've opened up https://github.com/ros2/ros2_documentation/pull/3604, which should fix the concerns you pointed out in the test.

With that, I'm going to consider this completed.  Thank you for testing!

closed 2023-05-15 by clalancette