osrf/ros2_test_cases #992: Recording A Bag From Your Own Node Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Recording A Bag From Your Own Node Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Recording A Bag From Your Own Node Cpp page](https://docs.ros.org/en/iron/Tutorials/Ros2bag/Recording-A-Bag-From-Your-Own-Node-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jmackay2, 2023-05-06:
I don't seem to have permission to assign myself this task.

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] Besides the bag format saying `sqlite3` instead of `mcap`, there weren't any obvious errors. The PR for this fix is [here](https://github.com/ros2/ros2_documentation/compare/iron...jmackay2:ros2_documentation:patch-2).

clalancette, 2023-05-06:
I went ahead and merged in your fix for the documentation; it is appreciated.

Thanks for testing!

closed 2023-05-06 by clalancette