osrf/ros2_test_cases #993: Recording A Bag From Your Own Node Py

Yadunund, 2023-05-01:
Check the documentation for the 'Recording A Bag From Your Own Node Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Recording A Bag From Your Own Node Py page](https://docs.ros.org/en/iron/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jacobperron, 2023-05-08:
This looks like a duplicate of https://github.com/osrf/ros2_test_cases/issues/994

Yadunund, 2023-05-11:
Nice catch.

closed 2023-05-11 by Yadunund