osrf/ros2_test_cases #994: Recording A Bag From Your Own Node Python

Yadunund, 2023-05-01:
Check the documentation for the 'Recording A Bag From Your Own Node Python' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Recording A Bag From Your Own Node Python page](https://docs.ros.org/en/iron/Tutorials/Ros2bag/Recording-A-Bag-From-Your-Own-Node-Python.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

hwoithe, 2023-05-04:
> The --dependencies argument will automatically add the necessary dependency lines to package.xml.txt``.

remove .txt``

> Because you used the --dependencies option during package creation, you don’t have to manually add dependencies to package.xml.txt``.

remove .txt``

> However, the writer requires serialised message to store in the bag.

missing article?

clalancette, 2023-05-04:
Good call on all counts.  See https://github.com/ros2/ros2_documentation/pull/3518 for fixes.

hwoithe, 2023-05-09:
I retested and it is working.

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

The only minor point I wanted to make is that it isn't clear why the line `self.subscription` is there. If I go by [this example](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html#id1), then it is used to "# prevent unused variable warning."

With that in mind, why doesn't the `self.timer` in the [bag_recorder_nodes_py ](https://docs.ros.org/en/iron/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-Py.html#write-a-python-node) example require a similar line?

clalancette, 2023-05-09:
> The only minor point I wanted to make is that it isn't clear why the line `self.subscription` is there. If I go by [this [...]

In both cases, you need to hold a handle to the resource you've created.  If you just call `create_subscription` without assigning it to anything, for instance, the subscription will be created and then immediately destroyed.

I believe that the `# prevent unused variable warning.` comment is there to prevent linters from (erroneously) thinking that this variable is unused.  Probably the `self.timer` one needs similar treatment, but I'm honestly not sure.

If you have ideas on how to improve the wording there, then please feel free to open an issue or a PR at https://github.com/ros2/ros2_documentation.

Otherwise, I'm going to consider this finished.  Thank you for the testing!

closed 2023-05-09 by clalancette

hwoithe, 2023-05-09:
> In both cases, you need to hold a handle to the resource you've created. If you just call create_subscription without assigning it to anything, for instance, the subscription will be created and then immediately destroyed. 

I agree with this regarding the subscriber and timer being assigned to self to ensure that they stay alive . I was referring to the lone "self.subscription" line (with no assignment). I could have been a bit more clear. :)

> I believe that the # prevent unused variable warning. comment is there to prevent linters from (erroneously) thinking that this variable is unused. Probably the self.timer one needs similar treatment, but I'm honestly not sure.

This is what I was going after and wondering about in the previous post. Why is it required to silence the linter for the subscriber example but not for the timer? The linter doesn't complain on my current configuration. I was hoping to learn the correct configuration so that we can be consistent across examples.

>Otherwise, I'm going to consider this finished. Thank you for the testing!

I agree, please close. Thanks much!