osrf/ros2_test_cases #995: Recording And Playing Back Data

Yadunund, 2023-05-01:
Check the documentation for the 'Recording And Playing Back Data' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Recording And Playing Back Data page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
Assign to me

tylerjw, 2023-05-03:
[image]

tylerjw, 2023-05-03:
Worked as expected. Unsure if we should do find/replace on "bag file" with "bag directory". Bagfile is a term I've heard people use and search for even if it is now a directory containing two files.

Yadunund, 2023-05-04:
> Worked as expected. Unsure if we should do find/replace on "bag file" with "bag directory". Bagfile is a term I've heard people use and search for even if it is now a directory containing two files.

Thanks for flagging this. I think it's worth to make the documentation clearer and have opened https://github.com/ros2/ros2_documentation/pull/3504

Thanks for testing this ticket! I'll consider this done

closed 2023-05-04 by Yadunund