osrf/ros2_test_cases #997: RQt Overview Usage

Yadunund, 2023-05-01:
Check the documentation for the 'RQt Overview Usage' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [RQt Overview Usage page](https://docs.ros.org/en/iron/Tutorials/RQt-Overview-Usage.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
:heavy_check_mark: Able to follow all steps in RQt Overview Usage page. 

[image]

[image]

clalancette, 2023-05-10:
Thanks for testing!

closed 2023-05-10 by clalancette