osrf/ros2_test_cases #999: Run 2 nodes in a single docker container

Yadunund, 2023-05-01:
Check the documentation for the 'Run 2 nodes in a single docker container' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Run 2 nodes in a single docker container page](https://docs.ros.org/en/iron/Tutorials/Run-2-nodes-in-a-single-docker-container.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

hwoithe, 2023-05-04:
$ docker --version
Docker version 23.0.2, build 569dd73
$ docker pull osrf/ros:iron-desktop
Error response from daemon: manifest for osrf/ros:iron-desktop not found: manifest unknown: manifest unknown

I performed the tests with osrf/ros:rolling-desktop instead of osrf/ros:iron-desktop.
I also replaced "docker-compose up" with "docker compose up" in the last section.

Otherwise, everything seems fine.

clalancette, 2023-05-04:
> I also replaced "docker-compose up" with "docker compose up" in the last section.

Thanks.  As I understand it, `docker-compose` is the older, V1 version of the command (which looks like it is going to [stop being supported in June](https://docs.docker.com/compose/)), while `docker compose` is the newer V2 version of the command.  So I've opened a PR to update this to `docker compose`: https://github.com/ros2/ros2_documentation/pull/3527

Since this is a minor documentation fix, and we have a PR open for it now, I'm going to close this out.  Thank you for testing!

closed 2023-05-04 by clalancette