osrf/ros2_test_cases #1042: Full System Testing: Nav2 Edition

SteveMacenski, 2023-05-03:
Someone please tag `amd` `debian` `cyclone` `jammy`. Should there be a second ticket if I'm also going to do `fastdds`? I'll likely test `binary` too but I'm not committing to that up front :-)

clalancette, 2023-05-03:
> Someone please tag `amd` `debian` `cyclone` `jammy`. Should there be a second ticket if I'm also going to do `fastdds`? I'll likely test `binary` too but I'm not committing to that up front :-)

Yeah, a separate ticket for fastdds would be preferred.  That way it is easier to make this a standard part of the testing in the future if we want to.

SteveMacenski, 2023-05-03:
- [ ] Localization localizes
- [ ] Gazebo Classic comes up, hooks into plugins, and stays up
- [ ] Rviz comes up and displays the data I would expect for all 30 visualization types
- [ ] BT navigator takes commands and timely cancels or preempts them
- [ ] Controller and Planner servers publish out costmaps at rate to the Behavior server 
- [ ] CLI commands for clearing costmap, commanding to point, and similar work
- [ ] Run a couple of canned demos in the `simple_commander` examples
- [ ] Rviz panel takes and sends commands without crashing the rviz window 
- [ ] laser scans and sensor data pass through custom callback groups to populate the main costmap
- [ ] TF trees are complete and publishing at a regular clip
- [ ] Pluginlib plugins are being appropriately identified and loaded
- [ ] Composition nodes are being identified and loaded into the component containers 
- [ ] Lifecycle nodes + services are bringing up the system into an active state and tearing down to a clean exit 
- [ ] Internal actions appear to be working as expected for smooth navigation

SteveMacenski, 2023-05-03:
@clalancette Ticket from Iron --> https://github.com/colcon/colcon-core/issues/554. Should I report this anywhere else?

SteveMacenski, 2023-05-03:
https://github.com/osrf/ros2_test_cases/issues/1044 is the Fast-DDS version (linking to be able to find later)

SteveMacenski, 2023-05-03:
Woof.  Nav2 appears to be broken but I don't think its Iron, though it could I suppose :shrug:. We're seeing Nav2 receiving AMCL localization service requests, but not honoring the content of them (and seems to randomly just bounce around) & Gazebo Classic appears to be producing data that doesn't seems sensible for where the robot truly is in the environment + reporting that the robot is outside of the local costmap bounds for update. This feels like TF, services, and/or gazebo issues which I doubt are related to the core packages but maybe some regression we have within Nav2 itself. This happened around the time of the last Rolling sync, but that could be entirely coincidental. 

Once its moving, the planning and control stack seems to be functional from really quick testing. 

I'm currently building a branch that I nuked all changes back to March 1 to see if it still is a problem - at which point it should be clear. I narrowed down from our CI results the last job I can clearly know it didn't happen in and the first when it appears. I can't see any commits in that range that touch any of this, but I flashed back much further than even those.

SteveMacenski, 2023-05-03:
OK, now I'm asking for some attention, I believe this is not Nav2 related and is instead ROS 2 core regressions. 

I've backed out all of the commits since last Christmas on Nav2 and I'm still seeing the same problem unchanged on Iron binaries.

I was able to find the last nightly build on Rolling that succeeded, and it was the night of April 13th (early morning hours of April 14th). This corresponds with the last Rolling release on the 14th before the Iron fork which broke Nav2's CI (`rosidl` issue, not relevant). Since the first build after resolving that CI issue, we see all system tests failing with this issue described above since. 

To confirm from a different direction, I then repeated this exercise using Humble and Nav2's current state without rolling back commits and things work perfectly (after a couple of 2 line changes for deprecated API). So the Iron vs Humble and Pre-last-Rolling-release shows that difference in behavior. So in summary, I think something in the last Rolling release before the fork broke some stuff that we see in Iron now. 

I scanned through the list of packages updated in the last sync and I see RMW, DDS, and client library stuff as well as TF2. Regressions here could explain the behavior. Nothing else in the list stood out to me, though I would have added Gazebo to that list if the sync included any updates there. 

I know this isn't very specific about where or what, but this is a flag worth raising early :flags:

SteveMacenski, 2023-05-03:
Followup - I went through the commits in the time range in rclcpp and TF2 but nothing stood out to me, so I took a more emperical approach.

Building Humble's version of `geometry2` (just selected something arbitrarily old but relatively new not to have API issues) in the Nav2 workspace seems to make the problems disappear. While Rviz and other tools are using the binary versions, Nav2 luckily is self contained enough that at least all the Nav2 modules can use that version of TF2 instead when built alongside it.

That seems to resolve the issues, which is welcome news that I don't need to dig into rclXXX or the RMW vendors. 

As best I can tell, there's only 2 commits that seem like they might be causing an issue that have been in the `ros2` branch after march:
- https://github.com/ros2/geometry2/commit/3bac620777043a1c8761cd74c1bb814667341b65 (and https://github.com/ros2/geometry2/commit/7afeb6c2b300be5a6b97eed0fa822a1d294e7b77)
- https://github.com/ros2/geometry2/commit/abfd97d7d7174207704b6cf2128c963e2b27bd04

I'm in the process of rebuilding geometry2 + nav2 without these to see if either is the culprit, but perhaps someone involved with those decisions (@clalancette) might have an educated guess looking at the recent commit history https://github.com/ros2/geometry2/commits/rolling

clalancette, 2023-05-04:
> OK, I've identified it. Compiling TF next to Nav2, I bisected commits from the last time I knew it was working properly for us `0.30.0` and now and found the behavior breaks when [ros2/geometry2@3bac620](https://github.com/ros2/geometry2/commit/3bac620777043a1c8761cd74c1bb814667341b65) is introduced (and by extension [ros2/geometry2@7afeb6c](https://github.com/ros2/geometry2/commit/7afeb6c2b300be5a6b97eed0fa822a1d294e7b77)).

Thanks for running this one to ground, that is really helpful.  I'll comment more on https://github.com/ros2/geometry2/issues/600

Yadunund, 2023-05-19:
@SteveMacenski is it safe to mark this test as a success?

SteveMacenski, 2023-05-19:
I will not be able to complete this. I spent the 2 days I had allocated for Iron testing on debugging the TF issue and unfortunately I need to be conscious of my time

Yadunund, 2023-05-19:
Understood. Thanks for update!

clalancette, 2023-06-23:
Some of this got tested, which is good.  Since we are out of the tutorial party now, I'm going to close this out as untested.

closed 2023-06-23 by clalancette