osrf/ros2_test_cases #1044: Full System Testing: Nav2 Edition

SteveMacenski, 2023-05-03:
Someone please tag amd debian fastdds jammy. 

- [ ] Localization localizes
- [ ] Gazebo Classic comes up, hooks into plugins, and stays up
- [ ] Rviz comes up and displays the data I would expect for all 30 visualization types
- [ ] BT navigator takes commands and timely cancels or preempts them
- [ ] Controller and Planner servers publish out costmaps at rate to the Behavior server 
- [ ] CLI commands for clearing costmap, commanding to point, and similar work
- [ ] Run a couple of canned demos in the `simple_commander` examples
- [ ] Rviz panel takes and sends commands without crashing the rviz window 
- [ ] laser scans and sensor data pass through custom callback groups to populate the main costmap
- [ ] TF trees are complete and publishing at a regular clip
- [ ] Pluginlib plugins are being appropriately identified and loaded
- [ ] Composition nodes are being identified and loaded into the component containers 
- [ ] Lifecycle nodes + services are bringing up the system into an active state and tearing down to a clean exit 
- [ ] Internal actions appear to be working as expected for smooth navigation

Yadunund, 2023-05-19:
@SteveMacenski any update on this test?

SteveMacenski, 2023-05-19:
I will not be able to complete this. I spent the 2 days I had allocated for Iron testing on debugging the TF issue and unfortunately I need to be conscious of my time

Yadunund, 2023-05-19:
Understood. Thanks for update! It was a critical fix.

clalancette, 2023-06-23:
Some of this got tested, which is good. Since we are out of the tutorial party now, I'm going to close this out as untested.

closed 2023-06-23 by clalancette