osrf/ros2_test_cases #1052: ros msgs

bai-ivan, 2023-05-06:
Hi,
wanted to play around with things, and i also saw that testing while not having the std_msgs fails bad, so decided to give them a go under windows. Can't add labels here, but a set of "binary" ; "fastdds"; "windows" would fit .

### Setup
- Binary install of ros2 iron under the testing party requirements.
- Installed colcon via pip.
- Cloned the messages repo in a blank workspace folder's src, using iron branch.
- Ran ```colcon build ``` in the ws folder.

### Links
[common_interfaces](https://github.com/ros2/common_interfaces/tree/iron)

### Output
```
C:\Users\User\test>colcon build
Starting >>> std_msgs
Starting >>> std_srvs
[Processing: std_msgs, std_srvs]
[Processing: std_msgs, std_srvs]
Finished <<< std_srvs [1min 7s]
[Processing: std_msgs]
Finished <<< std_msgs [1min 46s]
Starting >>> geometry_msgs
Starting >>> actionlib_msgs
[Processing: actionlib_msgs, geometry_msgs]
Finished <<< actionlib_msgs [50.6s]
[Processing: geometry_msgs]
Finished <<< geometry_msgs [1min 37s]
Starting >>> sensor_msgs
Starting >>> diagnostic_msgs
Starting >>> nav_msgs
Starting >>> shape_msgs
[Processing: diagnostic_msgs, nav_msgs, sensor_msgs, shape_msgs]
[Processing: diagnostic_msgs, nav_msgs, sensor_msgs, shape_msgs]
Finished <<< shape_msgs [1min 23s]
Starting >>> trajectory_msgs
Finished <<< diagnostic_msgs [1min 25s]
Finished <<< nav_msgs [1min 36s]
[Processing: sensor_msgs, trajectory_msgs]
Finished <<< trajectory_msgs [1min 7s]
Finished <<< sensor_msgs [2min 59s]
Starting >>> stereo_msgs
Starting >>> visualization_msgs
Starting >>> sensor_msgs_py
Finished <<< sensor_msgs_py [5.89s]
[Processing: stereo_msgs, visualization_msgs]
Finished <<< stereo_msgs [46.9s]
[Processing: visualization_msgs]
Finished <<< visualization_msgs [1min 28s]
Starting >>> common_interfaces
Finished <<< common_interfaces [14.0s]

Summary: 13 packages finished [8min 6s]

``` 

I will leave this open if someone want's me to test something else here. Will also try the dependency on packages issue with the output of this.

Yadunund, 2023-05-19:
@bai-ivan i'll close this since we determined that the issue with linking to std_msgs was because of a duplicate package name.

closed 2023-05-19 by Yadunund

kscottz, 2023-05-19:
@bai-ivan you are one of our top issue closers for the test and tutorial party! Will you please [fill out this short form](https://forms.gle/SDG2J3LJQTrtNAHH8) so we can send you some ROS Iron swag? Thank you so much!