osrf/ros2_test_cases #1064: Alternatives

marcoag, 2024-04-29:
Check the documentation for the 'Alternatives' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Alternatives page](https://docs.ros.org/en/jazzy/Installation/Alternatives.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-07:
@marcoag i can check this, please sign me up.

fujitatomoya, 2024-05-07:
since this issue is labeled with `Ubuntu Noble 24.04`, i did check the procedure https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html and https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html from https://docs.ros.org/en/jazzy/Installation/Alternatives.html

and, no problem observed.

fujitatomoya, 2024-05-07:
@marcoag we can close this.

closed 2024-05-07 by ahcorde