osrf/ros2_test_cases #1065: Latest Development Setup

marcoag, 2024-04-29:
Check the documentation for the 'Latest Development Setup' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Latest Development Setup page](https://docs.ros.org/en/jazzy/Installation/Alternatives/Latest-Development-Setup.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-07:
@marcoag i can check this.

fujitatomoya, 2024-05-07:
since this issue is labeled with `OS: Ubuntu Noble 24.04`, i will check the procedure https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html from https://docs.ros.org/en/jazzy/Installation/Alternatives/Latest-Development-Setup.html

fujitatomoya, 2024-05-07:
test result

```bash
root@51cdd59e1f3e:~/ros2_jazzy# . ~/ros2_jazzy/install/local_setup.bash
root@51cdd59e1f3e:~/ros2_jazzy# ros2 run demo_nodes_cpp talker
[INFO] [1715059953.359118865] [talker]: Publishing: 'Hello World: 1'
[INFO] [1715059954.359121749] [talker]: Publishing: 'Hello World: 2'
[INFO] [1715059955.359115616] [talker]: Publishing: 'Hello World: 3'
[INFO] [1715059956.358952638] [talker]: Publishing: 'Hello World: 4'
[INFO] [1715059957.359075915] [talker]: Publishing: 'Hello World: 5'

root@51cdd59e1f3e:~/ros2_jazzy# . ~/ros2_jazzy/install/local_setup.bash
root@51cdd59e1f3e:~/ros2_jazzy# ros2 run demo_nodes_py listener
[INFO] [1715059957.383835922] [listener]: I heard: [Hello World: 5]
[INFO] [1715059958.362153425] [listener]: I heard: [Hello World: 6]
[INFO] [1715059959.361986163] [listener]: I heard: [Hello World: 7]
[INFO] [1715059960.362094638] [listener]: I heard: [Hello World: 8]
```

and, clang build also completes without problems.

```bash
root@51cdd59e1f3e:~/ros2_jazzy# colcon build --cmake-force-configure
...
Finished <<< rviz_default_plugins [5min 17s]
Starting >>> rviz2
Finished <<< rviz2 [6.11s]

Summary: 363 packages finished [17min 11s]
```

we can close this one.

closed 2024-05-07 by ahcorde