osrf/ros2_test_cases #1066: RHEL Development Setup

marcoag, 2024-04-29:
Check the documentation for the 'RHEL Development Setup' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [RHEL Development Setup page](https://docs.ros.org/en/jazzy/Installation/Alternatives/RHEL-Development-Setup.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-03:
Wrongfully labeled "Ubuntu Noble 24.04" ?

clalancette, 2024-05-03:
> Wrongfully labeled "Ubuntu Noble 24.04" ?

Yeah, good point.  Will fix up the tags.

clalancette, 2024-05-14:
Step 1 - I was able to follow the documentation.  - Yep!
Step 2 - The documentation seemed clear to me.  - Yep!
Step 3 - The documentation didn't have any obvious errors. - Yep!

closed 2024-05-14 by clalancette