osrf/ros2_test_cases #1067: RHEL Install Binary

marcoag, 2024-04-29:
Check the documentation for the 'RHEL Install Binary' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [RHEL Install Binary page](https://docs.ros.org/en/jazzy/Installation/Alternatives/RHEL-Install-Binary.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-03:
Wrongfully labeled "Ubuntu Noble 24.04" ?

clalancette, 2024-05-03:
> Wrongfully labeled "Ubuntu Noble 24.04" ?

Yes, will fix this one up.

clalancette, 2024-05-14:
Step 1 - I was able to follow the documentation. - Yep!
Step 2 - The documentation seemed clear to me. - Yep!
Step 3 - The documentation didn't have any obvious errors. - There were a couple of errors related to rosdep; see https://github.com/ros2/ros2_documentation/pull/4451 for the fixes.

Given that we have the documentation PR open, going ahead and closing this one.

closed 2024-05-14 by clalancette