osrf/ros2_test_cases #1068: Ubuntu Development Setup

marcoag, 2024-04-29:
Check the documentation for the 'Ubuntu Development Setup' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Ubuntu Development Setup page](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-07:
@marcoag i can check this one too.

fujitatomoya, 2024-05-07:
@marcoag i think this test case is included within https://github.com/osrf/ros2_test_cases/issues/1065, so we already confirmed this.

closed 2024-05-07 by ahcorde