osrf/ros2_test_cases #1069: Ubuntu Install Binary

marcoag, 2024-04-29:
Check the documentation for the 'Ubuntu Install Binary' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Ubuntu Install Binary page](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-05:
Wrongfully labeled as "Build type: Debian"

marcoag, 2024-05-06:
> Wrongfully labeled as "Build type: Debian"

Updated, thanks!

clalancette, 2024-05-06:
I'm going to reopen this, as I don't think we had explicit confirmation that things were tested here.

mikaelarguedas, 2024-05-06:
Indeed I didn't test it :+1:

mikaelarguedas, 2024-05-06:
Is this different from https://github.com/osrf/ros2_test_cases/issues/1481 ?

If not I believe this can be closed (I tested it for all 3 RMW implementation and submitted some fixups here https://github.com/ros2/ros2_documentation/pull/4369)

marcoag, 2024-05-07:
Yup, I believe this is a duplicate but of https://github.com/osrf/ros2_test_cases/issues/1480 (FastDDS), will proceed to close again.

@clalancette feel free to reopen if there's something we missed here.

closed 2024-05-07 by marcoag