osrf/ros2_test_cases #1070: Windows Development Setup

marcoag, 2024-04-29:
Check the documentation for the 'Windows Development Setup' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Windows Development Setup page](https://docs.ros.org/en/jazzy/Installation/Alternatives/Windows-Development-Setup.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-04:
Wrongfully labeled "Ubuntu Noble 24.04"

marcoag, 2024-05-06:
> Wrongfully labeled "Ubuntu Noble 24.04"

True, changed that, thanks!

closed 2024-06-26 by marcoag