osrf/ros2_test_cases #1072: DDS Implementations

marcoag, 2024-04-29:
Check the documentation for the 'DDS Implementations' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [DDS Implementations page](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-06:
@marcoag i think this issue is to evaluate RTI connext dds debian packages?

fujitatomoya, 2024-05-06:
test result,

```bash
root@51cdd59e1f3e:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10FEB7D,0X8B07D2FD,0,0X204 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://55EB:7D00:0000:0000:0000:0000:0000:0000:7001
can't reach: locator: 17://1000:0000:591B:0000:55EB:7D00:0000:0000:1572911
[INFO] [1714969551.583590475] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714969552.583537959] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714969553.583579876] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714969554.583569184] [talker]: Publishing: 'Hello World: 4'
[INFO] [1714969555.583568272] [talker]: Publishing: 'Hello World: 5'

root@51cdd59e1f3e:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10FEB7D,0X8B07D2FD,0,0X204 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://55EB:7D00:0000:0000:0000:0000:0000:0000:7001
can't reach: locator: 17://1000:0000:591B:0000:55EB:7D00:0000:0000:1572911
[INFO] [1714969551.584707811] [listener]: I heard: [Hello World: 1]
[INFO] [1714969552.584379960] [listener]: I heard: [Hello World: 2]
[INFO] [1714969553.584491950] [listener]: I heard: [Hello World: 3]
[INFO] [1714969554.584502076] [listener]: I heard: [Hello World: 4]
[INFO] [1714969555.584480518] [listener]: I heard: [Hello World: 5]
```

once we hit `Ctrl-C` to the executables talker and listener, the following stack trace comes up.

```bash
root@51cdd59e1f3e:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10FEB7D,0X8B07D2FD,0,0X204 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://55EB:7D00:0000:0000:0000:0000:0000:0000:7001
can't reach: locator: 17://1000:0000:591B:0000:55EB:7D00:0000:0000:1572911
[INFO] [1714969551.583590475] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714969552.583537959] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714969553.583579876] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714969554.583569184] [talker]: Publishing: 'Hello World: 4'
[INFO] [1714969555.583568272] [talker]: Publishing: 'Hello World: 5'
^C[INFO] [1714969556.260873024] [rclcpp]: signal_handler(signum=2)
[D0200|DELETE Participant] PRESPsService_removeMatchingRemoteEndpoints:!precondition: "me == ((void *)0) || worker == ((void *)0) || remoteGuid == ((void *)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )"
        Backtrace:
        #1      ?? ??:0 [0xEE58D2E7]
        #2      ?? ??:0 [0xEE39DF72]
        #3      ?? ??:0 [0xEE37C127]
        #4      ?? ??:0 [0xEE31A650]
        #5      ?? ??:0 [0xEE31B97F]
        #6      ?? ??:0 [0xEE331C19]
        #7      ?? ??:0 [0xEE331801]
        #8      ?? ??:0 [0xEE30BE5F]
        #9      ?? ??:0 [0xEE373BDE]
        #10     ?? ??:0 [0xEF070541]
        #11     ?? ??:0 [0xEF03E1FC]
        #12     ?? ??:0 [0xEF92C9A5]
        #13     ?? ??:0 [0xEF962E0C]
        #14     ?? ??:0 [0xF0DE2833]
        #15     ?? ??:0 [0xF142DB40]
        #16     ?? ??:0 [0xF142DE68]
        #17     ?? ??:0 [0xF13EAB6A]
        #18     ?? ??:0 [0xF142D743]
        #19     ?? ??:0 [0xF142D7F1]
        #20     ?? ??:0 [0xF13EAAA6]
        #21     ?? ??:0 [0xF142D465]
        #22     ?? ??:0 [0x373A642A]
        #23     ?? ??:0 [0x373A50AA]
        #24     ?? ??:0 [0xF0E471CA]
        #25     ?? ??:0 [0xF0E4728B]
        #26     ?? ??:0 [0x373A6235]
[D0200|DELETE Participant] DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10FEB7D,0X8B07D2FD,0,0X1C1
```

i think this is not related to the procedure to be checked here.

fujitatomoya, 2024-05-06:
the issue is created to https://github.com/ros2/rmw_connextdds/issues/148.

i think we can close this test.

marcoag, 2024-05-06:
Sounds good, thanks!

closed 2024-05-06 by marcoag