osrf/ros2_test_cases #1073: Install Connext Security Plugins

marcoag, 2024-04-29:
Check the documentation for the 'Install Connext Security Plugins' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Install Connext Security Plugins page](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-Security-Plugins.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

samuel-ongzx, 2024-05-15:
Some comments:
- Requesting for a trial license & download the package provided in the [rti free trial page](https://www.rti.com/free-trial) sends you to download `v7.3.0`, with no mention of `v6.0.1` which is compatible with ROS2 Jazzy. I believe `v7.3.0.` doesn't work with Jazzy right now as mentioned [here](https://www.rti.com/free-trial/connext-ros2).
- Perhaps we should let people know how to download `v6.0.1` instead? 
- Also I found out from RTI that the evaluation license does include the security plugins (for `v7.3.0`)when the entire package is installed, no need for manual installation, mentioned [here](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-Security-Plugins.html). We do need to double check whether this is the case for `v6.0.1`.

[image]

samuel-ongzx, 2024-05-17:
## Configuration
- AMD64, Ubuntu 24.04, ROS2 Jazzy, ConnextDDS `v6.0.1`

## Feedback
Some proposed changes to the page since it is a little outdated, [page](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-Security-Plugins.html) is based on ConnextDDS `5.2.0`.

### Paragraph 1 [Original]
> The Connext DDS Libraries are included with ROS 2 under a [non-commercial license](https://www.rti.com/ncl) and do not include the security plug-in libraries. These libraries are available in the commercial, university and research license versions of RTI Connext DDS Pro, which is bundled with tools for system debugging, monitoring, record/replay, etc.

#### Proposed Edits
- `ok`

### Paragraph 2 [Original]
> The Connext DDS Evaluation Version (6.0.1) includes the security plugins, and can be downloaded via options available for [university, purchase or evaluation](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-University-Eval.html).

#### Proposed Edits
- Mention in `Install-Connext-University-Eval` [page](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-University-Eval.html) that the RTI trial page automatically requests & provides download for ConnextDDS `v7.3.0` which is not supported for ROS2. 
- The link to download ConnextDDS `v6.0.1` is [here](https://community.rti.com/page/rti-connext-dds-installers-ros-2).
- For other versions like `v6.1.0` which has prebuilt Arm64 binaries, we need to request it from RTI.

### Paragraph 3 [Original]
> A video walk-thru of this installation (tools and security plug-ins) is available [here](https://www.rti.com/gettingstarted/installwindows_secure) at the RTI website. The steps are:
#### Proposed Edits
- The link provided is outdated and just brings you to the general getting started page at RTI.
- Perhaps a better link for security plugins install would be [this](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/dds_security/html_files/RTI_SecurityPlugins_InstallationGuide/index.htm#SecurityPlugins/InstallationGuide/Installation.htm#gettingstartedwithsecurity_1183814207_1234790%3FTocPath%3D3.%2520Installation%2520Instructions%7C_____0).
- Extra link that references the main page for documentation is [here](https://community.rti.com/documentation/rti-connext-dds-601).
- Extra link that shows installation instructions for ConnextDDS `v6.0.1`, we can use this link here [this](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/getting_started/cpp11/before.html#installing-connext-dds).

### Paragraph 4 [Original]
> Install Connext DDS Pro (Host) This is a host-specific installer application (for Windows, Linux, MacOS) to install a ‘Host’ bundle which includes the Launcher, tools, and other software services. At the end of the installation, the RTI ‘Launcher’ program will be started. The Launcher is used to install target libraries, security plugins, and other layered services.
Use the Package Installer in Launcher
...
> The ‘RTI Package Installer’ is used to install ‘.rtipkg’ files – target libraries, security plug-ins, etc. Open the Package Installer and select all of the .rtipkg files that were included in the Connext DDS Secure bundle for installation:
Target Libraries - such as: rti_connext_dds-[version]-pro-target-[toolchain].rtipkg
Security Plugin Host - such as: rti_security_plugins-[version]-host-[toolchain].rtipkg
Security Plugin Target - such as: rti_security_plugins-[version]-target-[toolchain].rtipkg
OpenSSL Host - such as: openssl-1.0.2x-[version]-host-[toolchain].rtipkg
...
> Extract and Install OpenSSL This is included as an archive (.zip or otherwise) and can be simply extracted and copied to a convenient location on your host computer. As a suggestion, this could also be installed into the ‘rti_connext_dds-[version]’ directory in your home directory space (this was created during installation of the RTI host tools). Note: this directory location may need to be placed in your PATH environment variable. See the [RTI Security Plugins Getting Started Guide](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/dds_security/RTI_SecurityPlugins_GettingStarted.pdf) for more information.

#### Proposed Edits
- We should mention that the installation instructions for `evaluation` & `licensed` versions are different (this part was confusing to me).
- Security plugins & OpenSSL are installed be default in the `evaluation` version, and there is not need to install them manually. This is confirmed with from a correspondence from RTI. 
- However, for `licensed` RTI user, we do however need to install the security plugins manually as specified [here](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/dds_security/html_files/RTI_SecurityPlugins_InstallationGuide/index.htm#SecurityPlugins/InstallationGuide/Install_NonEval.htm#gettingstartedwithsecurity_1183814207_1289156%3FTocPath%3D3.%2520Installation%2520Instructions%7C_____2). 
- Original pdf link for the security plugin works too [RTI Security Plugins Getting Started Guide](https://community.rti.com/static/documentation/connext-dds/6.0.1/doc/manuals/connext_dds/dds_security/RTI_SecurityPlugins_GettingStarted.pdf).

## Others
Updated UI for installing `.rtipkg` packages (for security plugins) as a `licensed` user.
[image]

marcoag, 2024-06-26:
Thanks for the report here @samuel-ongzx! It would be great if you can add your improvements in the form of a PR so we can incorporate them into the documentation. Alternatively I will add this to the backlog and see if we have time to get to it. Thanks!

closed 2024-06-26 by marcoag

samuel-ongzx, 2024-06-27:
@marcoag  I will try to get to it this weekend!