osrf/ros2_test_cases #1075: Working with Eclipse CycloneDDS

marcoag, 2024-04-29:
Check the documentation for the 'Working with Eclipse CycloneDDS' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Working with Eclipse CycloneDDS page](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-07:
@marcoag i can take care of this.

btw, this issue is labeled with `Fast-DDS` but the task it to check `Cyclondds`... kind confused but complete verification as required anyway.

fujitatomoya, 2024-05-07:
test result,

```bash
root@51cdd59e1f3e:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp talker
[INFO] [1715062030.233967007] [talker]: Publishing: 'Hello World: 1'
[INFO] [1715062031.233933491] [talker]: Publishing: 'Hello World: 2'
[INFO] [1715062032.233930321] [talker]: Publishing: 'Hello World: 3'
[INFO] [1715062033.233934870] [talker]: Publishing: 'Hello World: 4'
[INFO] [1715062034.233930684] [talker]: Publishing: 'Hello World: 5'
[INFO] [1715062035.233942189] [talker]: Publishing: 'Hello World: 6'

root@51cdd59e1f3e:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp listener
[INFO] [1715062033.234413368] [listener]: I heard: [Hello World: 4]
[INFO] [1715062034.234402511] [listener]: I heard: [Hello World: 5]
[INFO] [1715062035.234396947] [listener]: I heard: [Hello World: 6]
```

no problem observed, we can close this one.

closed 2024-05-07 by ahcorde