osrf/ros2_test_cases #1078: Maintaining a Source Checkout

marcoag, 2024-04-29:
Check the documentation for the 'Maintaining a Source Checkout' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Maintaining a Source Checkout page](https://docs.ros.org/en/jazzy/Installation/Maintaining-a-Source-Checkout.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-06:
@marcoag i can check this, sign me up.

fujitatomoya, 2024-05-06:
No problem found, i think we can close this.

closed 2024-05-06 by ahcorde