osrf/ros2_test_cases #1079: RHEL Install RPMs

marcoag, 2024-04-29:
Check the documentation for the 'RHEL Install RPMs' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [RHEL Install RPMs page](https://docs.ros.org/en/jazzy/Installation/RHEL-Install-RPMs.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

clalancette, 2024-05-03:
> Wrongfully labeled "Ubuntu Noble 24.04" ?

Yep, will fix this one up.

clalancette, 2024-05-14:
> ### Step 1 - I was able to follow the documentation.

Yep!

> ### Step 2 - The documentation seemed clear to me.

Yep!

> ### Step 3 - The documentation didn't have any obvious errors.

Yep!