osrf/ros2_test_cases #1080: Testing

marcoag, 2024-04-29:
Check the documentation for the 'Testing' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Testing page](https://docs.ros.org/en/jazzy/Installation/Testing.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

ottojo, 2024-05-14:
I was about to look at this installation guide, but this issue may not be applicable to jazzy just yet?I noticed that in my ros:jazzy docker container, `/etc/apt/sources.list.d/ros2.list` does not seem to exist and `/etc/apt/sources.list.d/ros2-testing.list` contains the testing repo already. The debian install instructions also say to use the testing repositories for now.

I did briefly test the other two methods of using docker and the binary archives, which worked according to the documentation, although at this point those probably contain no newer versions than installed in `ros:jazzy` anyways. Feel free to let me know if this can be tested further.

## docker

A warning appeared when running the container or sourcing setup.bash:

```
[rti_connext_dds_cmake_module][warning] No RTI Connext DDS installation specified.. RTI Connext DDS will not be available at runtime,unless you already configured LD_LIBRARY_PATH manually.
```
but basic ros functionality (talker/listener) worked fine, and docker showed the image to be created 12h ago, so i guess the nightly-build also runs.

## binary archives

The same warning as above appears when sourcing the extracted `ros2-package-linux-x86_64.tar.bz2`. Binaries and nodes from archive can be used successfully (again verified with demo_nodes_cpp).

clalancette, 2024-05-14:
> I was about to look at this installation guide, but this issue may not be applicable to jazzy just yet?

Yes, you are exactly right.  During the beta/pre-release period, things are *only* available in the testing repositories.  So from that perspective, I would say that this is successful, so going ahead and closing this.  Thanks for testing!

closed 2024-05-14 by clalancette