osrf/ros2_test_cases #1082: Windows Install Binary

marcoag, 2024-04-29:
Check the documentation for the 'Windows Install Binary' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Windows Install Binary page](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

mikaelarguedas, 2024-05-04:
Wrongfully labeled "Ubuntu Noble 24.04" ?

marcoag, 2024-05-06:
> Wrongfully labeled "Ubuntu Noble 24.04" ?

Updated, thanks!

Trushant-Adeshara-UM, 2024-05-14:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-14:
Environment:

OS: Windows 11
Steps to Reproduce and Observations:

Step 1:

Followed the provided documentation with a slight variation in the setup: used OpenSSL 3.2.1, which was manually installed from source instead of using Choco. The process was straightforward and matched the documentation expectations.
Step 2:

The documentation clarity was satisfactory.
Step 3:

Encountered a warning related to the rti_connext_dds_cmake_module during the execution of local_setup.bat. This warning appears non-critical and was ignored for the continuation of the setup, as it did not affect the configuration process.
Outcome:

Both Publisher and Subscriber are operational as expected. Here is the snapshot demonstrating their functionality (please attach the snapshot or provide a link to it).
Conclusion:

The setup, following the deviation in OpenSSL version, is functional with minor warnings that do not impact the overall performance. Further clarity on handling such warnings in the documentation could be beneficial for future setups.

[image]