osrf/ros2_test_cases #1086: FastDDS Configuration

marcoag, 2024-04-29:
Check the documentation for the 'FastDDS Configuration' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [FastDDS Configuration page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/FastDDS-Configuration.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

samuel-ongzx, 2024-05-14:
## Process
- [x] Step 1 - I was able to follow the documentation.
- [x] Step 2 - The documentation seemed clear to me.
- [x] Step 3 - The documentation didn't have any obvious errors.

## Testing
- [x] Mixing synchronous and asynchronous publications in the same node ('ok' - setup, env setup, running, verification of output)
- [x] Using other FastDDS capabilities with XML (`ok` - limit subscribers, partitions, verification of output)
- [x] Configuring a service and a client ('ok' - ping service & client, verification of output)

Verification:
Tested `ok` - async & sync pub
[image]
Tested `ok` - ping
[image]

clalancette, 2024-05-14:
Fantastic, thank you for testing!

closed 2024-05-14 by clalancette