osrf/ros2_test_cases #1088: ROS2 Tracing Trace and Analyze

marcoag, 2024-04-29:
Check the documentation for the 'ROS2 Tracing Trace and Analyze' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [ROS2 Tracing Trace and Analyze page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/ROS2-Tracing-Trace-and-Analyze.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

samuel-ongzx, 2024-05-14:
## Process
- [x] Step 1 - I was able to follow the documentation.
- [x] Step 2 - The documentation seemed clear to me.
- [ ] Step 3 - The documentation didn't have any obvious errors.

`Step 3` not check as the `master` branch of [`performance_test`](https://gitlab.com/ApexAI/performance_test) does not support `jazzy`. Adding `jazzy` support in the [`CMakeLists.txt`](https://gitlab.com/ApexAI/performance_test/-/blob/master/performance_test/CMakeLists.txt?ref_type=heads#L500) passes compilation and execution. I have checked branch `2.1.2` and it does not have `jazzy` too.
`  elseif($ENV{ROS_DISTRO} MATCHES "jazzy")
    rosidl_get_typesupport_target(cpp_typesupport_target ${PROJECT_NAME} "rosidl_typesupport_cpp")
    target_link_libraries(${EXE_NAME} ${cpp_typesupport_target})`

## Test
- [ ] Installing and building (`ok` except for issue described above ^)
- [x] Tracking ('`ok' - tracing, and validating with babeltrace)
- [x] Analysis (`ok` - jazzy Jupyter notebook works)

## Verification
- Profiled node: `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py` using non-realtime kernel option
- Human readable output with `babeltrace ~/.ros/tracing/perf-test | less`
[image]

samuel-ongzx, 2024-05-15:
Attempting to add jazzy support to `performance_test`: https://gitlab.com/ApexAI/performance_test/-/merge_requests/399

samuel-ongzx, 2024-05-15:
PR has been merged by Erik! https://gitlab.com/ApexAI/performance_test/-/merge_requests/400
`Jazzy` support has been added into `performance_test` and everything works as it should!

clalancette, 2024-05-15:
> PR has been merged by Erik! https://gitlab.com/ApexAI/performance_test/-/merge_requests/400
> `Jazzy` support has been added into `performance_test` and everything works as it should!

Fantastic, thanks for being proactive and opening up the PR!  I'm going to go ahead and close this.

closed 2024-05-15 by clalancette