osrf/ros2_test_cases #1090: Recording A Bag From Your Own Node CPP

marcoag, 2024-04-29:
Check the documentation for the 'Recording A Bag From Your Own Node CPP' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Recording A Bag From Your Own Node CPP page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-CPP.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-04:
@marcoag i will check this, sign me up.

fujitatomoya, 2024-05-04:
this failed with the following error.

```bash
root@51cdd59e1f3e:~/ros2_ws# colcon build --packages-select bag_recorder_nodes
Starting >>> bag_recorder_nodes
--- stderr: bag_recorder_nodes
/root/ros2_ws/src/bag_recorder_nodes/src/data_generator_node.cpp: In constructor ‘DataGenerator::DataGenerator()’:
/root/ros2_ws/src/bag_recorder_nodes/src/data_generator_node.cpp:21:26: error: no matching function for call to ‘rosbag2_cpp::Writer::create_topic(<brace-enclosed initializer list>)’
   21 |     writer_->create_topic(
      |     ~~~~~~~~~~~~~~~~~~~~~^
   22 |       {"synthetic",
      |       ~~~~~~~~~~~~~
   23 |        "example_interfaces/msg/Int32",
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   24 |        rmw_get_serialization_format(),
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   25 |        ""});
      |        ~~~~
In file included from /root/ros2_ws/src/bag_recorder_nodes/src/data_generator_node.cpp:6:
/opt/ros/jazzy/include/rosbag2_cpp/rosbag2_cpp/writer.hpp:103:8: note: candidate: ‘void rosbag2_cpp::Writer::create_topic(const rosbag2_storage::TopicMetadata&)’
  103 |   void create_topic(const rosbag2_storage::TopicMetadata & topic_with_type);
      |        ^~~~~~~~~~~~
/opt/ros/jazzy/include/rosbag2_cpp/rosbag2_cpp/writer.hpp:103:60: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const rosbag2_storage::TopicMetadata&’
  103 |   void create_topic(const rosbag2_storage::TopicMetadata & topic_with_type);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/opt/ros/jazzy/include/rosbag2_cpp/rosbag2_cpp/writer.hpp:113:8: note: candidate: ‘void rosbag2_cpp::Writer::create_topic(const rosbag2_storage::TopicMetadata&, const rosbag2_storage::MessageDefinition&)’
  113 |   void create_topic(
      |        ^~~~~~~~~~~~
/opt/ros/jazzy/include/rosbag2_cpp/rosbag2_cpp/writer.hpp:113:8: note:   candidate expects 2 arguments, 1 provided
gmake[2]: *** [CMakeFiles/data_generator_node.dir/build.make:76: CMakeFiles/data_generator_node.dir/src/data_generator_node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/data_generator_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< bag_recorder_nodes [2.52s, exited with code 2]

Summary: 0 packages finished [2.62s]
  1 package failed: bag_recorder_nodes
  1 package had stderr output: bag_recorder_nodes
```

fujitatomoya, 2024-05-04:
@marcoag @clalancette https://github.com/ros2/ros2_documentation/pull/4358 can fix this problem.

after the fix, the procedure works with no problem. we can close this one once https://github.com/ros2/ros2_documentation/pull/4358 is merged.

closed 2024-05-06 by ahcorde