osrf/ros2_test_cases #1091: Recording A Bag From Your Own Node Py

marcoag, 2024-04-29:
Check the documentation for the 'Recording A Bag From Your Own Node Py' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Recording A Bag From Your Own Node Py page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-Py.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-04:
This tutorial is broken, the following error is detected.

```bash
root@51cdd59e1f3e:~/ros2_ws# ros2 run bag_recorder_nodes_py simple_bag_recorder
Traceback (most recent call last):
  File "/root/ros2_ws/install/bag_recorder_nodes_py/lib/bag_recorder_nodes_py/simple_bag_recorder", line 33, in <module>
    sys.exit(load_entry_point('bag-recorder-nodes-py==0.0.0', 'console_scripts', 'simple_bag_recorder')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/ros2_ws/install/bag_recorder_nodes_py/lib/python3.12/site-packages/bag_recorder_nodes_py/simple_bag_recorder.py", line 41, in main
    sbr = SimpleBagRecorder()
          ^^^^^^^^^^^^^^^^^^^
  File "/root/ros2_ws/install/bag_recorder_nodes_py/lib/python3.12/site-packages/bag_recorder_nodes_py/simple_bag_recorder.py", line 19, in __init__
    topic_info = rosbag2_py._storage.TopicMetadata(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. rosbag2_py._storage.TopicMetadata(id: int, name: str, type: str, serialization_format: str, offered_qos_profiles: List[rosbag2_py._storage.QoS] = [], type_description_hash: str = '')

Invoked with: kwargs: name='chatter', type='std_msgs/msg/String', serialization_format='cdr'
[ros2run]: Process exited with failure 1
```

this is generated by https://github.com/ros2/rosbag2/pull/1569.

fujitatomoya, 2024-05-04:
with https://github.com/ros2/ros2_documentation/pull/4357 fix for the documentation, this entire procedure works fine.

@marcoag after https://github.com/ros2/ros2_documentation/pull/4357 is merged, we can close this one.

closed 2024-05-04 by marcoag