osrf/ros2_test_cases #1094: Examine Traffic

marcoag, 2024-04-29:
Check the documentation for the 'Examine Traffic' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Examine Traffic page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Security/Examine-Traffic.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-16:
@marcoag i can check this, please sign me up for this.

fujitatomoya, 2024-05-16:
as described in https://docs.ros.org/en/jazzy/Tutorials/Advanced/Security/Examine-Traffic.html, we cannot examine network traffic with fast-dds, so i checked the followings.

- with fast-dds, we cannot see the network traffic data-plane. (we cannot see `Hello world` messages via UDP since it uses shared memory transport.)
- with cyclonedds, we can see the network traffic as it explains.

no problem observed, i think we can close this.

there is one minor fix required about the security enclaves keystore folder name.

```bash
root@tomoyafujita:~# ros2 run demo_nodes_cpp talker --ros-args --enclave /talker_listener/talker
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl: SECURITY ERROR: directory '/root/sros2_demo/demo_keys/enclaves/talker_listener/talker' does not exist., at ./src/rcl/security.c:202
[ros2run]: Aborted
```

this issue can be addressed by https://github.com/ros2/ros2_documentation/pull/4464

closed 2024-05-16 by ahcorde