osrf/ros2_test_cases #1095: Introducing ros2 security

marcoag, 2024-04-29:
Check the documentation for the 'Introducing ros2 security' page

## Configuration

* Chip set: AMD64
* Build type: Debian
* OS: Ubuntu Noble 24.04
* DDS: FastDDS

## Links

* [Introducing ros2 security page](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Security/Introducing-ros2-security.html)

## Process

    
            
### Step 1 - I was able to follow the documentation.
            
        

    
    

    

    
    
            
### Step 2 - The documentation seemed clear to me.
            
        

    
    

    

    
    
            
### Step 3 - The documentation didn't have any obvious errors.

fujitatomoya, 2024-05-16:
@marcoag i can check this, please sign me up.

fujitatomoya, 2024-05-16:
no problem observed, we can close this one.

<details>
  <summary>test result(click to expand!)</summary>

```shell
root@tomoyafujita:~# ros2 run demo_nodes_cpp talker --ros-args --enclave /talker_listener/talker
[INFO] [1715837234.055852308] [rcl]: Found security directory: /root/sros2_demo/demo_keystore/enclaves/talker_listener/talker
[INFO] [1715837235.068206868] [talker]: Publishing: 'Hello World: 1'
[INFO] [1715837236.068210240] [talker]: Publishing: 'Hello World: 2'
[INFO] [1715837237.068097546] [talker]: Publishing: 'Hello World: 3'
[INFO] [1715837238.068068894] [talker]: Publishing: 'Hello World: 4'
[INFO] [1715837239.068148126] [talker]: Publishing: 'Hello World: 5'
[INFO] [1715837240.068082475] [talker]: Publishing: 'Hello World: 6'
[INFO] [1715837241.068145002] [talker]: Publishing: 'Hello World: 7'
[INFO] [1715837242.068160498] [talker]: Publishing: 'Hello World: 8'
[INFO] [1715837243.068197903] [talker]: Publishing: 'Hello World: 9'
[INFO] [1715837244.068214952] [talker]: Publishing: 'Hello World: 10'
[INFO] [1715837245.068214394] [talker]: Publishing: 'Hello World: 11'
[INFO] [1715837246.068177686] [talker]: Publishing: 'Hello World: 12'
[INFO] [1715837247.068162485] [talker]: Publishing: 'Hello World: 13'
[INFO] [1715837248.068160367] [talker]: Publishing: 'Hello World: 14'
[INFO] [1715837249.068203406] [talker]: Publishing: 'Hello World: 15'
[INFO] [1715837250.068163435] [talker]: Publishing: 'Hello World: 16'
[INFO] [1715837251.068198205] [talker]: Publishing: 'Hello World: 17'
[INFO] [1715837252.068167589] [talker]: Publishing: 'Hello World: 18'
[INFO] [1715837253.068171552] [talker]: Publishing: 'Hello World: 19'
[INFO] [1715837254.068164228] [talker]: Publishing: 'Hello World: 20'
[INFO] [1715837255.068173498] [talker]: Publishing: 'Hello World: 21'

root@tomoyafujita:~# ros2 run demo_nodes_py listener --ros-args --enclave /talker_listener/listener
[INFO] [1715837238.063308827] [rcl]: Found security directory: /root/sros2_demo/demo_keystore/enclaves/talker_listener/listener
[INFO] [1715837241.079500645] [listener]: I heard: [Hello World: 7]
[INFO] [1715837242.070858432] [listener]: I heard: [Hello World: 8]
[INFO] [1715837243.070693977] [listener]: I heard: [Hello World: 9]
[INFO] [1715837244.070670724] [listener]: I heard: [Hello World: 10]
[INFO] [1715837245.070683571] [listener]: I heard: [Hello World: 11]
[INFO] [1715837246.070721025] [listener]: I heard: [Hello World: 12]
[INFO] [1715837247.070558755] [listener]: I heard: [Hello World: 13]
[INFO] [1715837248.070799926] [listener]: I heard: [Hello World: 14]
[INFO] [1715837249.070624845] [listener]: I heard: [Hello World: 15]
[INFO] [1715837250.070799016] [listener]: I heard: [Hello World: 16]
[INFO] [1715837251.070962067] [listener]: I heard: [Hello World: 17]
[INFO] [1715837252.070638914] [listener]: I heard: [Hello World: 18]
[INFO] [1715837253.070555367] [listener]: I heard: [Hello World: 19]
[INFO] [1715837254.070550735] [listener]: I heard: [Hello World: 20]
[INFO] [1715837255.070621226] [listener]: I heard: [Hello World: 21]

```

</details>

closed 2024-05-16 by ahcorde